VisorFolks/cyancore #42: Build Support on Windows OS

akashkollipara, 2021-10-06:
**Is your feature request related to a problem? Please describe.**
This feature is needed to build the cyancore software framework on Windows OS.
- As of now only Linux support is available but because of the paths, build fails even with windows supported toolchain.

**Describe the solution you'd like**
- N/A

**Describe alternatives you've considered**
- Use VSCode (Remote) with WSL enabled
- Use Ubuntu 20.04 Terminal
- Add VSCode plugin to support build and debug
- Bring in new build engine (!Last escort!)

akashkollipara, 2021-10-06:
Raised on behalf of @gkunalupta

pranjalchanda08, 2021-10-06:
We shall consider adding up extension of vs code to help people configure the build engine on cross platform setup.

pranjalchanda08, 2021-11-23:
hi @akashkollipara, Lets discuss upon this task

akashkollipara, 2021-11-25:
@gkunalupta please have a look at this

akashkollipara, 2022-01-09:
Hey @pranjalchanda08,

I want to say this is a closed issue!
People who queried about running makefile on windows were aswered to use wsl shell and proceed.

pranjalchanda08, 2022-01-09:
> Hey @pranjalchanda08,
> 
> I want to say this is a closed issue! People who queried about running makefile on windows were aswered to use wsl shell and proceed.

@akashkollipara,

That surely works as a workaround but on a longer run we need something that supports it on Win. I do remember we were discussing something on CDC to support this feature.

akashkollipara, 2022-01-09:
> 

Yes, but i am re-evaluating the efforts behind it...

pranjalchanda08, 2022-01-09:
Guess we can put it to lower priority.

closed 2022-02-07 by pranjalchanda08