VisorFolks/cyancore #49: Add light weight multi-tasking kernel in Terravisor

akashkollipara, 2021-10-14:
**Is your feature request related to a problem? Please describe.**
Terravisor needs multi-tasking support

**Describe the solution you'd like**
Create a mini kernel with various scheduling algorithms for multi tasking.

akashkollipara, 2021-12-21:
@pranjalchanda08 
Please get started. This will be a shared issue. We need to keep the status updated.

pranjalchanda08, 2021-12-21:
> @pranjalchanda08 Please get started. This will be a shared issue. We need to keep the status updated.

Sure

pranjalchanda08, 2022-12-27:
Update: 
1. Task Functions are all updated.
2. Semaphore Functions are updated
3. Supports 2 scheduling algorithm currently

To Be done:
1. Scheduler algorithm needs to be done
2. Need to have a function to recognize if the wait call is from Task or ISR to prevent locking.
3. Power save mech to be done.
4. Testing is to be done

pranjalchanda08, 2022-12-28:
Closing this issue with other open Items to the feature. Shall include them in Feature/Issue list in future.

closed 2022-12-28 by pranjalchanda08