VisorFolks/cyancore #101: FR: Add IS_ISR() API to depict an ISR context.

pranjalchanda08, 2022-02-06:
**Is your feature request related to a problem? Please describe.**
Add IS_ISR() API to depict an ISR context.

**Describe the solution you'd like**
An unified API to tell about current call context to help kernel being non-blocking if the kernel API is being called from ISR context.
 
```c
#if    (__ARM_ARCH_7A__      == 1U)
/* CPSR mode bitmasks */
#define CPSR_MODE_USER            0x10U
#define CPSR_MODE_SYSTEM          0x1FU

#define IS_IRQ_MODE()             ((__get_mode() != CPSR_MODE_USER) && (__get_mode() != CPSR_MODE_SYSTEM))
#else
#define IS_IRQ_MODE()             (__get_IPSR() != 0U)
#endif

#define IS_IRQ()                  IS_IRQ_MODE()
```

akashkollipara, 2022-04-14:
Added in_isr call in avr and riscv archs

akashkollipara, 2022-04-14:
Note: Our framework shouldn't be aware of the exception layer. The idea of the exception will be decided by the programmer during compile time. Inspired from RISC-V arch.

closed 2022-04-22 by akashkollipara