VisorFolks/cyancore #197: <CC_OS> Anti Deadlock system

pranjalchanda08, 2023-01-03:
**Is your feature request related to a problem? Please describe.**
The expectation is to detect and notify user about any thread not responding / deadlocked.

The Scheduler shall maintain a tick system logic to down-count a variable every single tick. 
The task is supposed to reset the ticks using a feed_api() and not let the timer ticks go down to absolute 0.

If the variable goes to 0 then it shall trigger a user callback to notify the thread in deadlock state and user can take a call to terminate the task if required. 

**A Psuedo OS watchdog. **

akashkollipara, 2023-01-03:
`Pseudo OS Watchdog` --> Software/Virtual WDog.

akashkollipara, 2023-01-03:
Alternatively... use hardware wdog to fire first IRQ to kill blocked task, blocked here refers to holding the resource for very long time or ends up waiting indefinitely for resource.

pranjalchanda08, 2023-01-03:
We do not want to do that as many MCUs do not have HW Watchdog or the user may want to use the watchdog in any other way.

pranjalchanda08, 2023-02-27:
Commits are merged. Hence closing the PR.

closed 2023-02-27 by pranjalchanda08