VisorFolks/cyancore #198: <CC_OS> Stack Overflow detection

pranjalchanda08, 2023-01-03:
**Is your feature request related to a problem? Please describe.**
To detect and notify task stack overflow using cannery methode.

akashkollipara, 2023-01-24:
Create a worker called as `housekeeper` in terravisor.
It should have similar arch as sysclk driver, basically calling via linked list.
When the kernel-driver is included in the build, the driver should auto add the housekeeper to list and it will be invoked by the housekeeper in sched upon `resched`.

This is my proposed design.

In that stack overflow is one of the kernel-drivers/features which should be enabled by `config.mk`.

akashkollipara, 2023-01-24:
The above design shall be extended to all visors.